stm32duino/VL53L8CX #2: Compilation error with Arduino IDE

cedrick-f, 2023-06-24:
Hi

I try to compile the example `VL53L8CX_HelloWorld_I2C.ino` into a Nucleo-F401RE board with Arduino IDE (version 2.0.4) on Windows 10 OS, but i get the following error during the compilation :

```
c:/users/xxx/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/12.2.1-1.2/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Cedrick\AppData\Local\Temp\arduino\sketches\2E654FFD584778FA4B92CD8319F2CC50/VL53L8CX_HelloWorld_I2C.ino.elf section '.rodata' will not fit in region 'FLASH'
c:/users/xxx/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/12.2.1-1.2/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: region 'FLASH' overflowed by 91800 bytes
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

thanks by advance for your help
Cédrick

fpistm, 2023-06-24:
This simply means you exceed the size of the flash. You can try to reduce your code or to optimize the core. You can refers to the wiki how to do tha.t. anyway maybe it cannot fit even with Optimization.

cedrick-f, 2023-06-24:
Among the STM32 MCU based boards, I selected Nucleo-64.
For information, the program compiles well if I choose Nucleo-144 (but of course cant't be uploaded)

fpistm, 2023-06-24:
You have to select also the part number. Else you build for the first one in the menu which is a C0 with 32k.

closed 2023-06-24 by fpistm