stm32duino/VL53L8CX #3: Add ESP32 to supported architectures?

PaulaScharf, 2024-02-09:
**Is your feature request/improvement related to a problem? Please describe.**

When this library is used on an ESP32 board, there is a warning, that it may be incompatible for this architecture.

**Describe the solution you'd like**

Add ESP32 to the supported architectures

**Additional context**

If tried it both on a custom microcontroller with an ESP32S2 processor and an UM Feather S2 board. Both work with this library. Would it be possible to add ESP32 to the supported architectures? The warning is quite confusing, when I show other people sketches that I made with this library.

Or at least it seems to be working. I only tried I2C communication. Are you aware of any shortcomings of this library on ESP32?

fpistm, 2024-02-09:
Hi @PaulaScharf 
We do not have esp32 so we could not test. 
Library should be 100% Arduino compatible so it probably works if other cores follow Arduino Core API.
If you have tested and it works then we could add it to the supported architecture.
Do not hesitate to provide a PR for this.

cparata, 2024-02-09:
Hi @PaulaScharf ,
I updated the library properties adding also the support for ESP32. You can download v1.0.3.
Best Regards,
Carlo

closed 2024-02-09 by cparata

PaulaScharf, 2024-02-09:
Nice, thanks for the quick resolve :D