stm32duino/VL53L8CX #4: ESP32 No I2C Communication

daniel-theia, 2024-06-13:
**Describe the bug**

No I2C response on the ESP32. The code is based on the following: https://github.com/stm32duino/VL53L8CX/blob/main/examples/VL53L8CX_HelloWorld_I2C/VL53L8CX_HelloWorld_I2C.ino

**To Reproduce**

Circuit layout:

[image]

Changes include the pinout:

```c++
#define I2C_RST_PIN -1
#define PWREN_PIN 32
#define GPIO1_PIN -1
#define GPIO2_PIN -1
#define LPN_PIN 26
#define SPI_I2C_N_PIN 27
#define SCL_PIN 33
#define SDA_PIN 25
#define NCS_PIN -1
#define MISO_PIN -1
```

The I2C/SPI selector is set to LOW:

```c++
  // Ground the SPI_I2C_N_PIN
  pinMode(SPI_I2C_N_PIN, OUTPUT);
  digitalWrite(SPI_I2C_N_PIN, LOW);
  delay(10);
```

Otherwise the code is pretty standard.

The read is failing , as tested here:

```c++
  unsigned long timeout = millis() + 1000;
  do {
    status = sensor_vl53l8cx_top.vl53l8cx_check_data_ready(&NewDataReady);
    if(millis() > timeout){
      SerialPort.println("Timeout whilst waiting for new data");
      break;
    }
  } while (!NewDataReady);
```

I'm also unable to detect the I2C address by scanning.

Steps to reproduce the behavior:
1. Compile
2. Run

**Expected behavior**

Ranging data is outputted.

**Screenshots**

```c++
Serial started.
I2C started.
VL53L8CX started.
VL53L8CX ranging...
```

And no data after.

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Arduino IDE version: 1.8.19
 - STM32 core version: N/A
 - Library version: 1.0.2
 - Tools menu settings if not the default: N/A
 - Upload method: Default serial

**Hardware (please complete the following information):**
 - Board Name: ESP32 Dev Board Node MCU
 - Hardware Revision: N/A
 - Extra hardware used if any: No

**Additional context**

cparata, 2024-06-14:
Hello @daniel-theia ,
first of all, I think that the I2C Pull-Up resistors are already mounted on the satellite, so you do not need to add them in your setup. Then, I think that SPI_I2C_N_PIN can be connected directly to GND in your setup. You must connect to the MCU the PWREN_PIN and put it HIGH like in the example code of the VL53L8CX library, otherwise the device is not powered up.
Best Regards,
Carlo

daniel-theia, 2024-06-14:
Thank you for your quick reply @cparata .

> first of all, I think that the I2C Pull-Up resistors are already mounted on the satellite, so you do not need to add them in your setup.

I tried with and without. Page 7 of the datasheet suggests 2.2k ohms on SDA and SCL, but I assume 4.7k ohms is okay with slightly less current and less drive current to overcome the pull-up? https://www.st.com/en/imaging-and-photonics-solutions/vl53l8cx.html

On the SATEL datasheet I am seeing the break-off board having 2.2k ohm pull-ups on page 6: https://www.st.com/en/ecosystems/satel-vl53l8.html#overview

> Then, I think that SPI_I2C_N_PIN can be connected directly to GND in your setup.

Does it matter if it is connected to ground or driven low? I'm seeing that the ESP32 can drive a pin up to 40mA. It would kind of defeat the purpose of having a power enable for an MCU if you need a large driving force to use it.

> You must connect to the MCU the PWREN_PIN and put it HIGH like in the example code of the VL53L8CX library, otherwise the device is not powered up.

My code is largely based on the example and I believe it does drive the pin high:

```c++
  #define PWREN_PIN 32 // NOTE: Defined above.

  // Enable PWREN pin if present
  if (PWREN_PIN >= 0) {
    pinMode(PWREN_PIN, OUTPUT);
    digitalWrite(PWREN_PIN, HIGH);
    delay(10);
  }
```

I'll try the following:

* Remove the I2C pull-ups (done already)
* Connect `SPI_I2C_N_PIN` directly to ground

Additionally I'll see what I can do about shortening the lines encase of additional capacitance. They are approximately 20cm and length matched to a good enough tolerance (+/- 10mm at worst), but we'll see. They are also soldered to remove the possibility of some jumper cable connection issue.

daniel-theia, 2024-06-18:
@cparata I've tried the following:

* Remove the I2C pull-ups
* Direct set `SPI_I2C_N_PIN`to ground
* Confirmed that `PWREN_PIN` is set high
* Re-soldered multiple times
* Switched to jumper headers
* Moved the I2C pins to the internal I2C pins (`SCL_PIN` on GPIO22 and `SDA_PIN` on GPIO21)

Next I will try to scope out the I2C lines, but I am running out of ideas. Any help is appreciated.

cparata, 2024-06-18:
Hello @daniel-theia ,
do you have the possibility to check the I2C signals with a logical analyzer? Could you share also the entire sketch that you are using?
Best Regards,
Carlo

daniel-theia, 2024-06-18:
[image]

@cparata  From what I can tell, the ACK is being set by the sensor. This is approximately what I would expect to see and the signals appear to be pulled well enough.

daniel-theia, 2024-06-18:
The code is (sorry for length of comment):

```c++
#include <Wire.h>
#include <vl53l8cx_class.h>

///* https://randomnerdtutorials.com/esp32-pinout-reference-gpios/ */
//#define I2C_RST_PIN -1 // (X)
//#define PWREN_PIN 18 // (11)
//#define GPIO1_PIN 4 // (4)
//#define GPIO2_PIN 5 // (5)
//#define LPN_PIN 2 // (38)
//#define SPI_I2C_N_PIN 8 // (12)
//#define SCL_PIN 38 // (31)
//#define SDA_PIN 39 // (32)
//#define NCS_PIN 3 // (11)
//#define MISO_PIN -1 // (X)

//// ESP8266
//#define I2C_RST_PIN -1
//#define PWREN_PIN D0
//#define GPIO1_PIN -1
//#define GPIO2_PIN -1
//#define LPN_PIN D2
//#define SPI_I2C_N_PIN D1
//#define SCL_PIN D3
//#define SDA_PIN D4
//#define NCS_PIN -1
//#define MISO_PIN -1

// https://www.waveshare.com/img/devkit/accBoard/NodeMCU-32S/NodeMCU-32S-details-3.jpg
#define I2C_RST_PIN -1
#define PWREN_PIN 32
#define GPIO1_PIN -1
#define GPIO2_PIN -1
#define LPN_PIN 26
#define SPI_I2C_N_PIN -1//27
#define SCL_PIN 22//33
#define SDA_PIN 21//25
#define NCS_PIN -1
#define MISO_PIN -1

#define DEV_I2C Wire
#define SerialPort Serial

void print_result(VL53L8CX_ResultsData *Result);
void clear_screen(void);
void handle_cmd(uint8_t cmd);
void display_commands_banner(void);

// Components.
VL53L8CX sensor_vl53l8cx_top(&DEV_I2C, LPN_PIN, I2C_RST_PIN);

bool EnableAmbient = false;
bool EnableSignal = false;
uint8_t res = VL53L8CX_RESOLUTION_4X4;
char report[256];

void setup() {
  // Set non-connected to be floating
  if(NCS_PIN >= 0) pinMode(NCS_PIN, INPUT);
  if(MISO_PIN >= 0) pinMode(MISO_PIN, INPUT);
  if(GPIO1_PIN >= 0) pinMode(GPIO1_PIN, INPUT);
  
  // Ground the SPI_I2C_N_PIN
  pinMode(SPI_I2C_N_PIN, OUTPUT);
  digitalWrite(SPI_I2C_N_PIN, LOW);
  delay(10);
  
  // Enable PWREN pin if present
[... 201 lines cut ...]
}

void handle_cmd(uint8_t cmd)
{
  switch (cmd)
  {
    case 'r':
      toggle_resolution();
      clear_screen();
      break;

    case 's':
      toggle_signal_and_ambient();
      clear_screen();
      break;

    case 'c':
      clear_screen();
      break;

    default:
      break;
  }
}
```

cparata, 2024-06-18:
Hello @daniel-theia ,
first of all I2C_RST_PIN and SPI_I2C_N_PIN are the same pins. So, if you want to pass it as parameter in the constructor, you need to connect it to the MCU and leave the management to the library. Alternatively, you can connect the pin to ground and pass it as -1 in the constructor of the library. The rest of the code seems fine to me. So, where the example hangs on your side?
Best Regards,
Carlo

daniel-theia, 2024-06-18:
Hello @cparata ,

> first of all I2C_RST_PIN and SPI_I2C_N_PIN are the same pins. So, if you want to pass it as parameter in the constructor, you need to connect it to the MCU and leave the management to the library.

I didn't realise this, but it wouldn't have made a difference anyway given I grounded `SPI_I2C_N_PIN` as suggested. I've updated this and no difference.

> Alternatively, you can connect the pin to ground and pass it as -1 in the constructor of the library.

This is how I just had it, but thought I would try allowing the library to control the reset.

> The rest of the code seems fine to me.

It's very similar to the example sketch, it should be okay if that code was okay.

> So, where the example hangs on your side?

It doesn't hang anywhere, I2C simply fails to communicate and it returns. If I turn the debugging up on the ESP32 I see:

```
[  4379][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
[  4388][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
[  4398][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
```

It definitely fails at `sensor_vl53l8cx_top.init_sensor()` as it returns a non-OK status.

Physically the circuit looks good, all fresh wires, all continuity tested.

The data on the scope even looks good:

[image]

I'm unsure where we go from here?

cparata, 2024-06-18:
Hello @daniel-theia ,
could you try a simple I2C scanner code like this one:

```
// SPDX-FileCopyrightText: 2023 Carter Nelson for Adafruit Industries
//
// SPDX-License-Identifier: MIT
// --------------------------------------
// i2c_scanner
//
// Modified from https://playground.arduino.cc/Main/I2cScanner/
// --------------------------------------

#include <Wire.h>

// Set I2C bus to use: Wire, Wire1, etc.
#define DEV_I2C Wire
#define I2C_RST_PIN -1
#define PWREN_PIN 32
#define LPN_PIN 26
#define SPI_I2C_N_PIN -1//27
#define SCL_PIN 22//33
#define SDA_PIN 21//25

void setup() {
  // Enable PWREN pin if present
  if (PWREN_PIN >= 0) {
    pinMode(PWREN_PIN, OUTPUT);
    digitalWrite(PWREN_PIN, HIGH);
    delay(10);
  }

  // Enable LPN pin if present
  if (LPN_PIN >= 0) {
    pinMode(LPN_PIN, OUTPUT);
    digitalWrite(LPN_PIN, HIGH);
    delay(10);
  }

  Serial.begin(9600);
  while (!Serial)
     delay(10);
  Serial.println("\nI2C Scanner");
    // Initialize I2C bus.
  if(SDA_PIN < 0 || SCL_PIN < 0){
    DEV_I2C.begin();
  }else{
    DEV_I2C.begin(SDA_PIN, SCL_PIN);
  }
  Serial.println("I2C started.");
}

void loop() {
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    DEV_I2C.beginTransmission(address);
    error = DEV_I2C.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4)
    {
      Serial.print("Unknown error at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.println(address,HEX);
    }
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

  delay(5000);           // wait 5 seconds for next scan
}
```

daniel-theia, 2024-06-18:
I can very clearly see the addresses being sent, but it picks up nothing on the ESP32:

[image]

```
Scanning...
No I2C devices found

Scanning...
No I2C devices found
```

I slightly modified the code to reflect `SPI_I2C_N_PIN` being connected to a pin:

```c++
// SPDX-FileCopyrightText: 2023 Carter Nelson for Adafruit Industries
//
// SPDX-License-Identifier: MIT
// --------------------------------------
// i2c_scanner
//
// Modified from https://playground.arduino.cc/Main/I2cScanner/
// --------------------------------------

#include <Wire.h>

// Set I2C bus to use: Wire, Wire1, etc.
#define DEV_I2C Wire
#define PWREN_PIN 32
#define LPN_PIN 26
#define SPI_I2C_N_PIN 27
#define SCL_PIN 22//33
#define SDA_PIN 21//25

void setup() {
  // Enable PWREN pin if present
  if (PWREN_PIN >= 0) {
    pinMode(PWREN_PIN, OUTPUT);
    digitalWrite(PWREN_PIN, HIGH);
    delay(10);
  }

  // Enable LPN pin if present
  if (LPN_PIN >= 0) {
    pinMode(LPN_PIN, OUTPUT);
    digitalWrite(LPN_PIN, HIGH);
    delay(10);
  }

  // Set I2C pin if present
  if (SPI_I2C_N_PIN >= 0) {
    pinMode(SPI_I2C_N_PIN, OUTPUT);
    digitalWrite(SPI_I2C_N_PIN, LOW);
    delay(10);
  }

  Serial.begin(9600);
  while (!Serial)
     delay(10);
  Serial.println("\nI2C Scanner");
  // Initialize I2C bus.
  if(SDA_PIN < 0 || SCL_PIN < 0){
    DEV_I2C.begin();
  }else{
    DEV_I2C.begin(SDA_PIN, SCL_PIN);
  }
  Serial.println("I2C started.");
}

void loop() {
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    DEV_I2C.beginTransmission(address);
    error = DEV_I2C.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4)
    {
      Serial.print("Unknown error at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.println(address,HEX);
    }
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

  delay(5000);           // wait 5 seconds for next scan
}
```

daniel-theia, 2024-06-18:
Those 2.2k pull-up resistors appear to be there and doing their job:

[image]

And I can confirm that the board is getting 3.3V on the 3.3V bus, and ground is connected to ground. I could try connecting the 5V to the 5V pin instead? It should bypass the regulators on the ESP32 encase there is some capacitance issues on the lines?

One thing I note is that on initialisation it sends the firmware: https://github.com/stm32duino/VL53L8CX/blob/57d3bb60e74527ffd2114849dc103680ae78a167/src/vl53l8cx_api.cpp#L327

Any bit flip in that entire process and we're likely dead in the water.

fpistm, 2024-06-18:
I2C frequency ?

daniel-theia, 2024-06-18:
Okaaaaaayyyy.... So I connected 5V to 5V and guess who spoke back:

```
Scanning...
I2C device found at address 0x29  !
done

Scanning...
I2C device found at address 0x29  !
done
```

cparata, 2024-06-18:
> Okaaaaaayyyy.... So I connected 5V to 5V and guess who spoke back:
> 
> ```
> Scanning...
> I2C device found at address 0x29  !
> done
> 
> Scanning...
> I2C device found at address 0x29  !
> done
> ```
If now you use the same hardware setup, does the VL53L8CX example work?

daniel-theia, 2024-06-18:
@cparata I think we can say it can work:

[image]

Thank you very much for your help!

In the example, it helps to replace `\n` with `\n\r`, and then the USB serial can be read with `screen /dev/ttyUSB0 115200`

cparata, 2024-06-18:
I think that we can close the issue. :-)

closed 2024-06-18 by cparata