stm32duino/VL53L8CX #5: Issue with VL53L8CX Sensor Library on Arduino Due

LelioG, 2024-06-20:
Hello,

I am contacting you because I am having an issue with the library associated with the VL53L8CX sensor. I am trying to make the SATEL-VL53L8 setup work directly with an Arduino Due, **without using** the P-NUCLEO-53L8A1 (plugging directly the SATEL to the Arduino). I assume that using an Arduino Due should work because there are parts of the code that are specifically tailored for the Due.
After numerous debugging attempts, I find myself stuck in the "RdMulti function" of the "platform.cpp" file, specifically in the loop that checks the I2C bus transmission status, as endTransmission consistently returns a status of 2 (received NACK on transmit of address).

The code is based on the following: https://github.com/stm32duino/VL53L8CX/blob/main/examples/VL53L8CX_HelloWorld_I2C/VL53L8CX_HelloWorld_I2C.ino
To reproduce : 

[image]

I used the following pinout : 

#define LPN_PIN 2
#define I2C_RST_PIN -1
#define PWREN_PIN 3

The loop is here : 

    do {
      p_platform->dev_i2c->beginTransmission((uint8_t)((p_platform->address >> 1) & 0x7F));

      // Target register address for transfer
      buffer[0] = (uint8_t)(RegisterAddress >> 8);
      buffer[1] = (uint8_t)(RegisterAddress & 0xFF);
      p_platform->dev_i2c->write(buffer, 2);

      status = p_platform->dev_i2c->endTransmission(false);

      // Fix for some STM32 boards
      // Reinitialize the i2c bus with the default parameters
      #ifdef ARDUINO_ARCH_STM32
      if (status) {
        p_platform->dev_i2c->end();
        p_platform->dev_i2c->begin();
      }
#endif
      // End of fix
    }

Desktop :

OS: Windows 11 / Ubuntu 22.04
Arduino IDE version: 2.3.3
STM32 core version: N/A
Library version: 1.0.4
Tools menu settings if not the default: N/A
Upload method: Default serial

Hardware (please complete the following information):

Board Name: Arduino Due ( Atmel ATSAM3X8EAU ARM )
Hardware Revision: N/A
Extra hardware used if any: No

cparata, 2024-06-20:
Hello @LelioG ,
which I2C instance are you used for the Arduino DUE? In the [VL53L8CX_HelloWorld_I2C.ino](https://github.com/stm32duino/VL53L8CX/blob/main/examples/VL53L8CX_HelloWorld_I2C/VL53L8CX_HelloWorld_I2C.ino) we use by default Wire1 that is connected to D20 and D21. Could you try to connect SDA and SCL to those pins?
Best Regards,
Carlo

LelioG, 2024-06-20:
Hello @cparata ,
It is already connected it's the green and red lines as shown at the bottom right picture on the DUE.
Best regards,
Lelio

cparata, 2024-06-20:
Hello @LelioG ,
yes I know. I'm just wondering if for the pins that you are using, you must define Wire instead of Wire1.
Could you try to adjust the define?
Best Regards,
Carlo

cparata, 2024-06-20:
Ok I noticed now that the pin should be the same of Wire1 on your setup. I have a working setup with the Arduino DUE and the X-NUCLEO-53L8A1. The only difference between my setup and yours should be that I'm using the I2C pins on the top right side of the board and probably I use 3V3 instead of 5V. Could you try also to connect the 3V3 pin instead of 5V pin as power?
Best Regards,
Carlo

LelioG, 2024-06-21:
Hello,

The setup I used does not involve the X-NUCLEO-53L8A1. I am simply trying to connect the SATEL-VL53L8 directly to the Arduino Due. I have already tried powering it with 3.3V, but the Arduino does not detect it, while it does detect the SATEL-VL53L8 at 5V.

Additionally, I have checked the sensor on the I2C bus and found that its address is indeed 0x29. 
Best regards,

Lelio

cparata, 2024-06-21:
Hello @LelioG ,
actually the X-NUCLEO-53L8A1 mounts the same sensor of the SATEL-VL53L8, so if the library works with the X-NUCLEO, it should work also with the SATEL-VL53L8. The only difference is the wire connections with the 2 boards. Anyway, if the I2C scanner works on your hardware setup, I don't see any issue to get it working also with the VL53L8CX library. Could you share the entire your sketch?
Best Regards,
Carlo

LelioG, 2024-06-21:
Hello @cparata ,

I tested another code and I can see the SATEL, connect to it, and the I2C connection works. However, the problem arises when using the VL53L8CX library, where I get stuck in the following path:

init_sensor() -> vl53l8cx_is_alive() -> RdByte() -> RdMulti()

Then I arrive at this loop in RdMulti:

```
do {
  p_platform->dev_i2c->beginTransmission((uint8_t)((p_platform->address >> 1) & 0x7F));

  // Target register address for transfer
  buffer[0] = (uint8_t)(RegisterAddress >> 8);
  buffer[1] = (uint8_t)(RegisterAddress & 0xFF);
  p_platform->dev_i2c->write(buffer, 2);

  status = p_platform->dev_i2c->endTransmission(false);

  // Fix for some STM32 boards
  // Reinitialize the i2c bus with the default parameters
  #ifdef ARDUINO_ARCH_STM32
  if (status) {
    p_platform->dev_i2c->end();
    p_platform->dev_i2c->begin();
  }
  #endif
  // End of fix
} 

```

And I remain stuck in this loop.

Best regards,

Lelio

cparata, 2024-06-21:
Ok @LelioG ,
next week I will try to reproduce your setup and I will come back to you.
Best Regards,
Carlo

cparata, 2024-06-24:
Hi @LelioG ,
your setup works on my side but, as said in a previous message, you must use the "Wire" instance of the I2C instead of "Wire1".
You can find below the sketch that I used. The hardware setup is the same that you showed in the first picture.

```
 /**
 ******************************************************************************
 * @file    VL53L8CX_HelloWorld_I2C.ino
 * @author  STMicroelectronics
 * @version V1.0.0
 * @date    12 June 2023
 * @brief   Arduino test application for STMicroelectronics VL53L8CX
 *          proximity sensor satellite based on FlightSense.
 *          This application makes use of C++ classes obtained from the C
 *          components' drivers.
 ******************************************************************************
 * @attention
 *
 * <h2><center>&copy; COPYRIGHT(c) 2021 STMicroelectronics</center></h2>
 *
 * Redistribution and use in source and binary forms, with or without modification,
 * are permitted provided that the following conditions are met:
 *   1. Redistributions of source code must retain the above copyright notice,
 *      this list of conditions and the following disclaimer.
 *   2. Redistributions in binary form must reproduce the above copyright notice,
 *      this list of conditions and the following disclaimer in the documentation
 *      and/or other materials provided with the distribution.
 *   3. Neither the name of STMicroelectronics nor the names of its contributors
 *      may be used to endorse or promote products derived from this software
 *      without specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
 * AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
 * IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
 * DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE
 * FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL
 * DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR
 * SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER
 * CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY,
 * OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
 * OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
 *
 ******************************************************************************
 */
/*
 * To use these examples you need to connect the VL53L8CX satellite sensor directly to the Nucleo board with wires as explained below:
 * pin 1 (SPI_I2C_n) of the VL53L8CX satellite connected to pin GND of the Nucleo board
 * pin 2 (LPn) of the VL53L8CX satellite connected to pin A3 of the Nucleo board
 * pin 3 (NCS) not connected
 * pin 4 (MISO) not connected
 * pin 5 (MOSI_SDA) of the VL53L8CX satellite connected to pin D14 (SDA) of the Nucleo board
 * pin 6 (MCLK_SCL) of the VL53L8CX satellite connected to pin D15 (SCL) of the Nucleo board
 * pin 7 (PWREN) of the VL53L8CX satellite connected to pin D11 of the Nucleo board
 * pin 8 (I0VDD) of the VL53L8CX satellite not connected
 * pin 9 (3V3) of the VL53L8CX satellite connected to 3V3 of the Nucleo board
 * pin 10 (1V8) of the VL53L8CX satellite not connected
 * pin 11 (5V) of the VL53L8CX satellite not connected 
 * GPIO1 of VL53L8CX satellite connected to A2 pin of the Nucleo board (not used)
 * GND of the VL53L8CX satellite connected to GND of the Nucleo board
 */
 
/* Includes ------------------------------------------------------------------*/
#include <Arduino.h>
#include <Wire.h>
#include <vl53l8cx_class.h>

#ifdef ARDUINO_SAM_DUE
  #define DEV_I2C Wire
#else
  #define DEV_I2C Wire
[... 215 lines cut ...]
{
  switch (cmd)
  {
    case 'r':
      toggle_resolution();
      clear_screen();
      break;

    case 's':
      toggle_signal_and_ambient();
      clear_screen();
      break;

    case 'c':
      clear_screen();
      break;

    default:
      break;
  }
}
```

Best Regards,
Carlo

cparata, 2024-06-24:
This is the output that I can see with Arduino DUE

[image]

LelioG, 2024-06-24:
Hello @cparata,
Thank you very much for your help, that was the problem.
Best regards,
Lelio

closed 2024-06-24 by LelioG