stm32duino/VL53L8CX #8: No Serial output from Nucleo F401RE when trying to get data from SATEL VL53L8

yaswanthsk04, 2024-08-04:
**Describe the bug**
I am trying to run VL53L8CX_HelloWorld_I2C.ino on Nucleo-F401RE with satel VL53L8 breakout board. But no output in serial monitor.

the program hangs in the init function.
[image]

**To Reproduce**

Steps to reproduce the behavior:
1). Connections as per instructions from Github repo
2). Installed latest stm32duino boards
3). Installed Library from arduino Lib manager
4). Compile and Build 
5). Open serial Monitor with baud rate 460800
6). No output in serial monitor

**Expected behavior**
Expecting Output of sensor data in serial monitor

**Code Used to debug:**
```
#include <vl53l8cx.h>

#ifdef ARDUINO_SAM_DUE
  #define DEV_I2C Wire1
#else
  #define DEV_I2C Wire
#endif
#define SerialPort Serial

#define LPN_PIN A3
#define PWREN_PIN 11

void print_result(VL53L8CX_ResultsData *Result);
void clear_screen(void);
void handle_cmd(uint8_t cmd);
void display_commands_banner(void);

// Components.
VL53L8CX sensor_vl53l8cx_top(&DEV_I2C, LPN_PIN);

bool EnableAmbient = false;
bool EnableSignal = false;
uint8_t res = VL53L8CX_RESOLUTION_4X4;
char report[256];
uint8_t status;

/* Setup ---------------------------------------------------------------------*/
void setup()
{
 SerialPort.begin(460800);
  SerialPort.println("Entered Setup");
  // Enable PWREN pin if present
  if (PWREN_PIN >= 0) {
    pinMode(PWREN_PIN, OUTPUT);
    digitalWrite(PWREN_PIN, HIGH);
    delay(10);
  }
  SerialPort.println("Finished Setting up PWREN");
  

  // Initialize I2C bus.
  DEV_I2C.begin();
  SerialPort.println("Finished Setting up I2C Begin");
  // Configure VL53L8CX component.
  sensor_vl53l8cx_top.begin();
  SerialPort.println("Finished Setting up Sensor Begin");
  status = sensor_vl53l8cx_top.init();
  SerialPort.println("Finished Setting up Sensor Init");

[... 248 lines cut ...]
```
**Finished Read Block...** is missing.

This is where my knowledge and experience ends. I was not able to debug more about the VL53L8CX_RdByte() func and its return value.

**Screenshots**
[image]

**Desktop:**
 - OS: Windows 11
 - Arduino IDE version: 2.3.2
 - STM32 core version: 2.8.1
 - Library version: 2.0.0
 - Tools menu settings if not the default: Board-> Nucleo-64; Board Part Number-> Nucleo-F401RE; 
 - Upload method: SWD
[image]

**Hardware:**
 - Board Name:Nucleo F401RE
 - Hardware Revision: Rev C
 - Extra hardware used if any: SATEL VL53L8 TOF sensor Breakout Board

**Additional Info**
Uploading the Library file as a zip file along with the wrapper function.
[STM32duino_VL53L8CX.zip](https://github.com/user-attachments/files/16487646/STM32duino_VL53L8CX.zip)

cparata, 2024-08-05:
Hello @yaswanthsk04 ,
first of all, I would check the I2C connectivity. Could you run the following code with your hardware setup and see if the sensor is able to respond on the I2C bus?

```
// SPDX-FileCopyrightText: 2023 Carter Nelson for Adafruit Industries
//
// SPDX-License-Identifier: MIT
// --------------------------------------
// i2c_scanner
//
// Modified from https://playground.arduino.cc/Main/I2cScanner/
// --------------------------------------

#include <Wire.h>

// Set I2C bus to use: Wire, Wire1, etc.
#define DEV_I2C Wire
#define LPN_PIN A3
#define PWREN_PIN 11

void setup() {
  // Enable PWREN pin if present
  if (PWREN_PIN >= 0) {
    pinMode(PWREN_PIN, OUTPUT);
    digitalWrite(PWREN_PIN, HIGH);
    delay(10);
  }

  // Enable LPN pin if present
  if (LPN_PIN >= 0) {
    pinMode(LPN_PIN, OUTPUT);
    digitalWrite(LPN_PIN, HIGH);
    delay(10);
  }

  Serial.begin(9600);
  while (!Serial)
     delay(10);
  Serial.println("\nI2C Scanner");
  // Initialize I2C bus.
  DEV_I2C.begin();
  Serial.println("I2C started.");
}

void loop() {
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    DEV_I2C.beginTransmission(address);
    error = DEV_I2C.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4)
    {
      Serial.print("Unknown error at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.println(address,HEX);
    }
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

  delay(5000);           // wait 5 seconds for next scan
}
```

yaswanthsk04, 2024-08-05:
Hi @cparata:
Thank you for your reply.

The I2C Scanner output is follows:

```
Scanning...
No I2C devices Found
```

Thanks,
Yaswanthkumar

cparata, 2024-08-05:
Hello @yaswanthsk04 ,
it means that there is an issue in your hardware setup. You should discover the ToF sensor with 0x29 I2C address. Until you don't get it, it does not make sense to run the VL53L8 library.
Please, check your wire connections.
Best Regards,
Carlo

cparata, 2024-08-05:
Hi @yaswanthsk04 ,
The connections in the guide should be fine. I checked the picture of your hardware setup and it seems fine to me even if I'm not able to see clearly all the connections. Please, be sure that SPI_I2C_N pin is connected to GND and check also the SCL pin that seems to me not well connected on the board.
Best Regards,
Carlo

yaswanthsk04, 2024-08-05:
Hello @cparata:

Thank you for your reply.

based on the following Image from Application Note of ST Electronics on **How to connect VL53L8 to a nucleo board** I found that, the sensor must be powered using 5V pin instead of 3.3v pin

[image]

Link to the Document: https://www.st.com/resource/en/application_note/an5945-how-to-connect-the-satelvl53l8-to-an-stm32-nucleo64-board-stmicroelectronics.pdf

Connecting the sensot to 5V pin works fine. I am able to find the I2C address as 0x29.

yaswanthsk04, 2024-08-05:
Hello @cparata:

Yes, it is working now!!!

:)

[image]

Did you connect it on 3.3v?

did it work for you in 3.3v?

cparata, 2024-08-05:
Hello @yaswanthsk04 ,
Unfortunately, I don't have the possibility to check. I will update the README and the comments in the library to be aligned with the User Manual.
Thanks for testing it.
Best Regards,
Carlo

cparata, 2024-08-05:
I released a new version 2.0.1 fixing the comments in the README and in the comments inside the examples.
I close this issue.
Best Regards,
Carlo

closed 2024-08-05 by cparata