stm32duino/VL53L8CX #9: ESP32: VL53L8CX_DISABLE_xxxx macros cause reset

LateNightIceCream, 2024-08-10:
**Describe the bug**
The `VL53L8CX_DISABLE_xxxx` defines in `platform_config_custom.h` / `plaform_config_default.h` do not work and it to automatically reset after starting.

**To Reproduce**

In the following code, I simply try to send a field of the ranging result to Serial:

`main.cpp`:
```c++
#include <Arduino.h>
#include "vl53l8cx.h"

#define SDA_PIN 11
#define SCL_PIN 7
#define LPn 18 
#define CS_PIN 12 
#define SPI_PIN 21

#define DEV_I2C Wire

VL53L8CX sensor(&DEV_I2C, LPn);

const uint8_t ranging_frequency_hz = 15;
const uint8_t resolution = VL53L8CX_RESOLUTION_8X8;
uint8_t status;

void setup() {
  Serial.begin(460800);
  delay(500);

  pinMode(CS_PIN, OUTPUT);
  digitalWrite(CS_PIN, LOW);

  pinMode(SPI_PIN, OUTPUT);
  digitalWrite(SPI_PIN, LOW);

  DEV_I2C.setClock(400000);
  DEV_I2C.begin(SDA_PIN, SCL_PIN);
  Serial.println("> Started I2C");

  sensor.begin();
  Serial.println("> Started sensor");

  status = sensor.init();
  Serial.println("> Initialized sensor");
  Serial.print("Status: ");
  Serial.println(status);

  status = sensor.set_resolution(resolution);
  status = sensor.set_ranging_mode(VL53L8CX_RANGING_MODE_CONTINUOUS);
  status = sensor.set_ranging_frequency_hz(ranging_frequency_hz);
  Serial.println("> Configured sensor");
  Serial.print("Status: ");
  Serial.println(status);

  status = sensor.start_ranging();
  Serial.println("> Started ranging");
  Serial.print("Status: ");
  Serial.println(status);
}

void loop() {
  VL53L8CX_ResultsData Results;
  uint8_t NewDataReady = 0;

  do {
    status = sensor.check_data_ready(&NewDataReady);
  } while (!NewDataReady);

  if ((!status) && (NewDataReady != 0)) {
    status = sensor.get_ranging_data(&Results);

    for (int i = 0; i < resolution; i++) {
      Serial.print(Results.signal_per_spad[i]);
      if (i < (resolution - 1)) {
        Serial.print(",");
      }
    }
    Serial.println();
  }
  delay(10);
}
```

`platform_config_custom.h`: (example)
```c++
#ifndef _PLATFORM_CONFIG_CUSTOM_H_
#define _PLATFORM_CONFIG_CUSTOM_H_

#ifndef VL53L8CX_NB_TARGET_PER_ZONE
  #define   VL53L8CX_NB_TARGET_PER_ZONE   1U
#endif

#define VL53L8CX_DISABLE_MOTION_INDICATOR

#endif
```

Steps to reproduce the behavior:
1. Define one or more of the `VL53L8CX_DISABLE_xxxx` macros in `platform_config_custom.h` 
2. Compile + upload
3. Monitor Serial output

**Expected behavior**
`platform_config_default.h` says I can define the below values to disable certain unneeded functionality:

https://github.com/stm32duino/VL53L8CX/blob/3535c7d182c290f37ca9191557f574e08c0f5386/src/platform_config_default.h#L67-L81

My posted code works fine as long as none of the mentioned defines are defined. 
Once I define any of the above macros, my ESP32 board will continuously reset. I tried every one
of them. Sometimes, the code does get to the "> Initialized sensor" line in setup but then it still resets.

**Desktop (please complete the following information):**
 - OS: NixOS (Linux 6.6.42)
 - PlatformIO Core, version 6.1.15
 - Library version: 2.0.1
 - Upload method: USB / Serial

**Hardware (please complete the following information):**
 - Board Name: WEMOS S2 Mini (ESP32 S2)
 - Hardware Revision: V1.0.0
 - Extra hardware used if any: [Pololu VL53L8CX Carrier](https://www.pololu.com/product/3419)

fpistm, 2024-08-13:
Hi @LateNightIceCream 
did you test using Arduino IDE ?

LateNightIceCream, 2024-08-13:
Hey! I just installed Arduino IDE on a Windows machine and used the code I provided above and... it actually works! I can define all of the macros without problems. I don't know what went wrong with platformio...

fpistm, 2024-08-13:
Thanks for the feedback. We support only Arduino not Pio. So I close this issue. You should file an issue on pio GitHub or on the forum.

closed 2024-08-13 by fpistm