stm32duino/VL53L8CX #10: SPIClass error in VL53L8CX_HelloWorld_SPI.ino

2hgeng, 2024-08-29:
- Code: VL53L8CX_HelloWorld_SPI.ino 
- Compiling the code in Arduino IDE ver 2.3.2
- Using Arduino Portenta H7 lite (https://store-usa.arduino.cc/products/portenta-h7-lite?gad_source=1&gclid=CjwKCAjwuMC2BhA7EiwAmJKRrOc5p091CSuoDOuGgptWaQMRpb22Qh0VMhP6ngdd5oX9LtIufg3TNRoCIecQAvD_BwE)
- Board package: Arduino Mbed OS Portenta Boards ver 4.1.5
- Library STM32duino VL53L8CX ver 2.0.1

Compilation error:
error: no matching function for call to 'arduino::HardwareSPI::HardwareSPI(unsigned int, unsigned int, unsigned int)'
 SPIClass DEV_SPI(SPI_MOSI_PIN, SPI_MISO_PIN, SPI_CLK_PIN);

cparata, 2024-08-30:
Hi @2hgeng ,
it seems that for Arduino mbedOS derived board there is not the constructor for the SPIClass as you can see [here](https://github.com/arduino/ArduinoCore-API/blob/master/api/HardwareSPI.h). In your specific case, you can try to use directly the SPI instance and see if it works.
So, try to replace:

`SPIClass DEV_SPI(SPI_MOSI_PIN, SPI_MISO_PIN, SPI_CLK_PIN);`

with

`#define DEV_SPI SPI`

Pay attention to connect the VL53L8CX SPI pins to the pins of the SPI instance on the Portenta board.

Best Regards,
Carlo

2hgeng, 2024-08-30:
I am connecting to SPI1 of Portenta board, the start ranging function status is 255 (decimal) and no data is received from sensor.

`#define CS_PIN PI_0
#define DEV_SPI SPI

VL53L8CX sensor_vl53l8cx_top(&DEV_SPI, CS_PIN);

void setup()
{
// Initialize SPI bus.
  DEV_SPI.begin();

  // Set SPI settings
  DEV_SPI.beginTransaction(SPISettings(1000000, MSBFIRST, SPI_MODE1));

  // Configure VL53L8CX component.
  sensor_vl53l8cx_top.begin();
  status = sensor_vl53l8cx_top.init();

  // Start Measurements
  status = sensor_vl53l8cx_top.start_ranging();
}`

cparata, 2024-08-30:
Hi @2hgeng ,
could you share your hardware setup? Which board with VL53L8CX are you using?
Best Regards,
Carlo

2hgeng, 2024-08-30:
The board is Arduino Portenta H7 lite sitting on the breakout board.

https://store-usa.arduino.cc/products/portenta-h7-lite?gad_source=1&gclid=Cj0KCQjw28W2BhC7ARIsAPerrcI-NFLuPr318Kd2GvifCJs0zx6EXtQp5hI4ihMdr3MSMh_EKonwCWkaAovoEALw_wcB

Breakout board:
https://store-usa.arduino.cc/products/arduino-portenta-breakout?gad_source=1&gclid=Cj0KCQjw28W2BhC7ARIsAPerrcJUZICk2R0fy3FeLyzWbwocanHVzHs9hNcBy6Cb8Xfg37AcwincDbAaAn__EALw_wcB

cparata, 2024-08-30:
And how the VL53L8CX sensor is connected to the Portenta board?

2hgeng, 2024-08-30:
I am using VL53L8CX breakout board
https://www.pololu.com/product/3419

and connecting it to Portenta breakout board SPI1.

[image]

cparata, 2024-08-30:
Did you connect the SPI_I2C_N pin of the Polulu board at 3.3V?

2hgeng, 2024-08-30:
Yes it is at 3.3V, the Polulu board by default pulls it high.

cparata, 2024-09-02:
Hi @2hgeng ,
are you sure about it? Giving a look at the schematics of the Pololu VL53L8CX board, it seems that the SPI_I2C_N pin is floating.
Best Regards,
Carlo
[image]

cparata, 2024-09-03:
Hi @2hgeng ,
any update on this topic? Did you try to connect the SPI_I2C_N pin to Vdd in your hardware setup? It should be one of the 4 floating pins in the picture that you attached above.
Best Regards,
Carlo

2hgeng, 2024-09-03:
Yes, it is connected to 3.3V power

cparata, 2024-09-03:
Could you share a picture of your hardware setup and the software sketch that you are using?

2hgeng, 2024-09-06:
Looks like the Pololu sensor breakout board is using a level shifter which is an issue. I am trying to get new board with new shifter and see if that works. I will keep you posted. Thank you.

closed 2025-05-13 by fpistm