stm32duino/VL53L8CX #12: build fails when instanciating 2 differents ToF (L7 and L8)

jmchiappa, 2025-05-08:
When instantiating these 2 different libraries VL53L8CX and VL53L7CX, the build fails.

It comes from that some header files have the same name in these libraries. So that, the reentrance guard prevents from parsing the second file of the other library.
And the build fails because some definitions can not be found.

**Describe the solution you'd like**
I suggest to add a prefix on "common" files : 
```
- platform.h                -> vl53l8cx_platform.h
- platform_config.h         -> vl53l8cx_platform_config.h
- platform_config_default.h -> vl53l8cx_platform_config_defaulth
- platform.cpp              -> vl53l8cx_platform.cpp
```
All #include should also be updated.

`union Block_header` seems to be common to both libraries, but are the same. So maybe add a definition like : 
```
#ifndef BLOCK_HEADER
#define BLOCK_HEADER
union Block_header {
  uint32_t bytes;
  struct {
    uint32_t type : 4;
    uint32_t size : 12;
    uint32_t idx : 16;
  };
};
#endif
```

Otherwise I strongly suggest to use namespace that will prevent from overlapping some common definitions.

By the way, I could submit a PR if needed.

BR

cparata, 2025-05-09:
Hi @jmchiappa,
thanks to have submitted this issue. I agree with you; probably the simplest solution is to use the #ifndef guard in the common struct. Your contribution is welcome, so you can submit the PR.
Thanks in advance and Best Regards,
Carlo

jmchiappa, 2025-05-09:
Hi @cparata ,
Thanks for your reply.
PR are submitted in both repositories.
Let me know.
Jean-Marc